MitjaNemec/ReplicateLayout #2: Selecting Remove existing tracks/zones/drawings deletes items from source layout

sslupsky, 2022-02-01:
Noticed an issue today where if the "Remove existing tracks/zones/drawings" is checked, some items in the source layout are removed.  The destination layout includes the missing items.

Here are a couple screenshots that illustrate the issue.  Note in the second illustration, the GND tracks and via's are missing as well as the AUXOUT stub.

Source layout before replications:
[image]

Source layout after replication:
[image]

Destination layout after replication:
[image]

MitjaNemec, 2022-02-01:
Hi, thanks for reporting. Can you share this project? Or can you at least supply similar project, where you can recreate this issue?

sslupsky, 2022-02-01:
Here you go.
[replicate_issue_2.zip](https://github.com/MitjaNemec/ReplicateLayout/files/7982036/replicate_issue_2.zip)

MitjaNemec, 2022-02-02:
OK, I can replicate the issue. But it will take some time until I find the bug

closed 2022-02-03 by MitjaNemec

MitjaNemec, 2022-02-03:
Found the source of the bug, and fixed it. The bug was present also in 5.1.x release and it was present for quite a while. Again thanks for spotting it and reporting it.

But the fix will take a while to be published in PCM. You can download the latest release from Github and install it from file

sslupsky, 2022-02-03:
That is great.  Thank you.

sslupsky, 2022-02-04:
I downloaded the update.  It's actually worse now ... all the nets disappear and the zones as well in the source layout.  I'll update my layout and upload again so you can reproduce.

sslupsky, 2022-02-04:
I reverted to 1.0.1 and the problem seems the same.  Perhaps my update did not work correctly.  I'll let you know if I figure this out.

commit 6bb7ee82 (2022-02-03): Don't delete tracks outside bounding box if on common nets

Fixes #2