MitjaNemec/ReplicateLayout #4: Test project doesn't have libraries included and has special footprints

EeliK, 2022-02-03:
I can't start the test project layout from scratch because symbols refer to footprints which aren't in the KiCad libraries. Footprint libraries should be included or only standard footprints used.

MitjaNemec, 2022-02-03:
Can you elaborate a bit, why would you want to modify test project.

With the introduction of the PCM the preferred way to install the plugin is through PCM either through KiCad's package repository or one can install the package from the .zip archive accessible under releases.

EeliK, 2022-02-03:
For testing, of course. Just for convenience so that I don't have to write another project for testing the plugin. I wanted to try the whole workflow, starting from where KiCad puts the footprints when the first update from the schematic is done, so I deleted everything from the layout first.

I found out that the problem can be solved by using File -> Export -> footprints to new library, updating library links, and then updating the schematic from pcb. It would be better in principle to include libraries in public projects. But I can do testing now, so I don't really care.