MitjaNemec/ReplicateLayout #5: plugin finds wrong wx Frame

EeliK, 2022-02-03:
1. Open KiCad
2. Open eeschema
3. Open pcbnew
4. Tools -> plugins -> replicate
5. Eeschema window is focused on
6. Message window on top of eeschema

When the plugin is run the next time, even when there's something to replicate and a footprint selected, the plugin main window raises eeschema.

I added logging after `logger.info("Frame repr: " + repr(self.frame))` in action_replicate_layout.py -> Run().

`logger.info("Frame: " + self.frame.GetTitle())`

The result:

`02-03 23:37:33 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 304:Frame: replicate_layout_test_project [replicate_layout_test_project/] — Schematic Editor`

EeliK, 2022-02-03:
Possible solution:

```
windows = wx.GetTopLevelWindows()
for w in windows: print(w.GetName())

KicadFrame
PcbFrame
KiPython
SchematicFrame
ModEditFrame
```

Haven't we discussed about this while we made changes for 5.99? Anyway, I'll try to make this work.

EeliK, 2022-02-03:
OK, combining algorithm from the 5.99 version with w.GetName() works:

```
    def Run(self):
        # grab PCB editor frame
        self.frame = wx.FindWindowById(pcbnew.ID_V_TOOLBAR).GetParent()
        topwindows = wx.GetTopLevelWindows()
        self.frame = [x for x in topwindows if 'PcbFrame' == x.GetName()][0]
```

(The old code is still there.) ID_V_TOOLBAR is used in other windows, too, and apparently wx returns the first one in the list.

MitjaNemec, 2022-02-04:
Thanks for finding this one. Can you write down which OS are you running.

Yeah, the old code was brittle as pcbnew or pcb editor frame name kept changing. and there were some issues on macOc. So I thought I was smart to refactor this to find the frame by looking up the parent of vertical toolbar. It looks like, I'll have to refactor it again. But I want to find out a cross platform solution, so it might take a while

EeliK, 2022-02-04:
I'm on Linux now but I can test on Windows if needed.

However, I don't think there's need to test further. I just grepped "PcbFrame" in the KiCad source and found out that 
```

./scripting/kicad_pyshell/__init__.py:        # frame names are "SchematicFrame" and "PcbFrame"
./scripting/kicad_pyshell/__init__.py:        frame = wx.FindWindowByName( "PcbFrame" )
```

so there seems to be even simpler solution which is already used within KiCad. :smile:

EeliK, 2022-02-05:
wx.FindWindowByName( "PcbFrame" ) works on Windows. I didn't actually test it on Linux yet, I have no aceess to my Linux machine ATM.

MitjaNemec, 2022-02-05:
Thanks for the info. It works on linux (Kubuntu 20.04) also, and I hope it will work in macOs. But as I can not get the handler for PCB editor vertical toolbar reliably, I'll have to show the plugin centered on PCB editor window.