MitjaNemec/ReplicateLayout #8: Error while loading plugin - ModuleNotFoundError

Rogash1, 2022-02-10:
Hi,
I cloned the newest release 1.0.3 of the ReplicateLayout Plugin into my "user\Documents\KiCad\6.0\scripting\plugins\" folder and can't get it to run. Whenever I try to run it, I get an error window telling me to check the place_footprints_error.log.
[image]
There is no error log with this name inside the plugin folder, just the replicate_layout_error.log

This logfile says "ModuleNotFoundError("No module named 'ReplicateLayout.replicate_layout_GUI'")"

Is this a bug or am I missing something?

Kind regards

Version: KiCad 6.0.1, OS: Windows 10

MitjaNemec, 2022-02-10:
With KiCad V6, the correct way to install the plugin is to download one (most commonly the latest) release archive.

The files in the repo are just a source code. All the derived code and data are generated and not included in the repository by default. So you are missing the .png files for icons which are generated from .svg files and .py files for GUI which are generated using wxFormBuilder

MitjaNemec, 2022-02-10:
If you have installed the plugin via PCM and manually, there might be some conflicts as PCM installs in one location and manualy you can install in another location. You should remove one installation

Rogash1, 2022-02-10:
Thanks for your quick reply. With the already-built release of version 1.0.3, the error does not occur anymore.
I guess the issue can be closed or deleted now. Thank you very much.

closed 2022-02-10 by MitjaNemec