MitjaNemec/ReplicateLayout #10: Unhelpful error message: "Property not found: Sheetfile"

Xasin, 2022-02-16:
Hi! 

I would absolutely love to use this plugin to help my KiCAD workflow along a little, and I've heard interest from multiple other people.
However, I am immediately faced with an issue that seems to prevent the plugin from working at all.

I want to replicate the PCB design of a KiCAD sheet that is replicated four times, by inserting a KiCAD subsheet that points to the same physical file. 
This works without issue on the PCB - the components are placed four times, with their IDs starting with a different number (I.e. C101, C201, C301 etc. referring to different instances of the same capacitor)

When it comes to replicating the PCB layout of these components however, the plugin immediately fails by throwing the error message "Property not found: Sheetfile"

This appears to be the same message as if choosing a PCB component that is not in a Subsheet (i.e. it's in the root sheet).

The error message gives no further instructions in regards to possible causes and fixes for this error message, and due to the absence of a README file or Wiki page elaborating on the exact requirements of this plugin, I am left with opening a GitHub Issue.

I will attach screenshots of the schematic in question:

[image]
[image]
[image]

Xasin, 2022-02-16:
One slight noteworthy change:
The project uses a main file called "swirl.kicad_sch" with the main PCB file being called "swirl.kicad_pcb" 

It appears the code is looking for a file ending in ".sch" (judging by this line: 
https://github.com/MitjaNemec/ReplicateLayout/blob/1ff11b62a3f6477b62fbf3aa0e04495f61567d66/replicate_layout.py#L95 )

As such, this might be a naming incompatibility.

MitjaNemec, 2022-02-16:
Thanks for reporting the bug.

The line in question is a remainder from the V5 version of the plugin. V6 version of the plugin does parse schematic files.

Is your project by chance a project which was started with KiCad V5? If it is, can you update layout from schematics and try running the plugin again.

In any case I'd really appreciate detailed error message (screenshot will do) and if you can attach `replicate_layout.log`file which should be in the project folder it should make my job a a lot easier

Xasin, 2022-02-16:
Hm - you're right, it appears to be an issue with projects started within KiCAD 5
I ran a very quick and dirty test on a new project generated purely in KiCAD 6, and it worked there.

However, updating the layout from the schematic does not appear to work. I created an entirely new item (an extra LED), which I imported into the PCB via the "Update PCB from Schematic" option.
This new LED did also not work as anchor-point for the "Replicate Layout" plugin.

The error log is quite small, so I'll insert it here:

```
02-16 19:06:17 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 322:Preparing replicator with D202 as a reference
02-16 19:06:17 com_github_MitjaNemec_ReplicateLayout.replicate_layout 99:getting a list of all footprints on board
02-16 19:06:17 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 329:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "/home/xasin/.var/app/org.kicad.KiCad/data/kicad/6.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/action_replicate_layout.py", line 326, in Run
    replicator = Replicator(board)
  File "/home/xasin/.var/app/org.kicad.KiCad/data/kicad/6.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 107, in __init__
    sheet_file = fp.GetProperty('Sheetfile')
  File "/app/lib/python3.9/site-packages/pcbnew.py", line 17716, in GetProperty
    raise KeyError("Property not found: " + key)
KeyError: 'Property not found: Sheetfile'
```

Completely removing the previous kicad_pcb file, creating a new one and reimporting everything fixes it.
Copying over the old PCB elements re-created the issue, so I began trying to isolate where exactly the issue occurs.

I think I found the source:

Your plugin is incompatible with the `KiBuzzard` custom font label generator plugin.
This plugin creates PCB components that have no corresponding schematic symbol, and exist purely on the PCB. This might cause your plugin to fail to parse the Sheetfile property, as there is none that exists.

I recommend simply ignoring components that have no sheetfile property, and filtering them out, or adding a dialog box that says "Following components have no sheetfile property and cannot be replicated:"

Crashing with no message given why the crash happened is a bit of an issue.

I can work without KiBuzzard for now, and the plugin seems to work fine without!
But I still recommend fixing up the error.

SebastianGrau, 2022-10-10:
> I think I found the source:
>
> Your plugin is incompatible with the `KiBuzzard` custom font label generator plugin. This plugin creates PCB components that have no corresponding schematic symbol, and [...]

I can confirm the same error and also the observation, that once _KiBuzzard_ is installed, the _Replicate Layout_ and _Save/Restore Layout_ plugins stop working and throw an error about the missing _Sheetfile_ property.