MitjaNemec/ReplicateLayout #9: Question: Can layouts be saved to a file?

greenscreenflicker, 2022-02-16:
I want to design custom building blocks ("blueprints"), for example for the LDO with two capacitors.
I want to save the design to be later replicated, on another design. Can that be done?
Thanks,
Michael

MitjaNemec, 2022-02-16:
Not at the moment

For V5 I had Save/Restore plugin that implemented this functionality, but I am still working on a V6 version of this. I am really in the early stages, as V6 schematics file format changed and if we want this to work properly sot that the layout is in sync with schematics, it is not as simple as taking some code of Replicate Layout plugin

MitjaNemec, 2022-09-15:
Closing as save restore plugin is now available (https://github.com/MitjaNemec/SaveRestoreLayout)

closed 2022-09-15 by MitjaNemec