MitjaNemec/ReplicateLayout #12: [Kicad issue] Hierarchical sheets and highlight on PCB issues

roykrikke, 2022-05-10:
First I want to share that your plugins for Kicad are great! Thank you for your effort. I wanted to point you to the issue below in Kicad. I suspect that this could potentially lead to changes in your plugins in the future.

https://gitlab.com/kicad/code/kicad/-/issues/11493

roykrikke, 2022-05-10:
Hallo,

I tried to share the issue on "command "select items on PCB" not working if executed on a deeper nested subsheet". I suspect that this could potentially lead to changes in your plugins in the future. I assumed that the plugin could use the possibility of nested sub-sheets and that is issue might have an impact.

MitjaNemec, 2022-05-10:
Yeah, I am not using that codepath. Thanks for the notice anyway. Sooner or later there is going to be a bug in a code that plugins use.

As for the nested hierarchy, the plugin already covers this.

closed 2022-05-10 by MitjaNemec