MitjaNemec/ReplicateLayout #14: Inclusion of non-hierarchical parts in Error

eewiz, 2022-05-20:
I'm using KiCAD 6.04.
I have one top sheet with a power supply and I/O connected to 16 identical hierarchical sections.
I completed the layout of section 1 of 16.
I used the plugin to replicate that first section into 4 out of 16 hierarchical sheets.
Each 4 duplicates are rotated by 90 degrees and I had not yet placed guide parts 5-16.
I had to deselect/select any checkbox to actually get the list of 16 sections to appear.
I set the check boxes as desired then I deselected sections 5-16 (all but 2-4) and hit OK.
Replicating 1 to 4 worked flawlessly. I was excited to realize how quickly I would be able to complete the remaining 12 sections.

I set 4 more guide parts (5-8) which are rotated 90.
I select the master part in section 1.
I run again and select to replicate sections 5-8 and I get this result.
[image]

Q222 is a transistor in the power supply on the top sheet of the schematic.
I checked... the schematic and PCB were in sync.
Actually it was Q2 to start with, but I renamed it from Q2 to Q222, did a resync, and the error changed from Q2 to Q222.
Q2 (Q222) is rotated 30 on the board, so I simply deleted it on the schematic, resynced and reran the plugin.
Same error, except now the message refers to R6, which is another non-hierarchical part in the power supply on the top sheet of the schematic.
Ctrl-Z, Ctrl-Z and Q222 is back.
It looks like it is trying to replicate non-hierarchical parts.
I've tried to selecting the master part from sections 2, 3 or 4 to replicate to sections 5-8. No good.
I've lain all 12 guide parts with no rotations in a clear space on the PCB and tried. No Good.

Any help would be greatly appreciated.
Thank You

MitjaNemec, 2022-05-20:
Thanks for the feedback.

There are a couple of things we can do to track down what is actually causing the issue.

1. Can you share KiCad project and attach it here?
2. If 1 is no go, can you restart the process from the start, and after each run of the plugin, grab the `replicate_layout.log` which should reside in the project folder and attach them here (after editing for sensitive information). Please denote precisely which .log is from which action.

MitjaNemec, 2022-05-20:
Additionally:

3. Can you look into the `kicad_pcb` file and look at their "path" fields. The should match.

eewiz, 2022-05-21:
Hello.
I discovered a method, more or less by accident, that worked for me.
I went backward in backups until I reached the one that had only one completed block before I successfully cloned it into blocks 2-4. 
I opened the project and set 15 guide parts out in free space.
I tried, still no go, same error.
It occurred to me to deleted a segment in each of the four tracks entering the master block to totally isolate the block from the rest of the board. These four tracks are visible in the lower left corner of the screen shot above.
I tried again and was immediately struck by the difference. Now, only the master block was highlighted.
In all previous attempts, the plugin would highlight ALL of the tracks on the board.
I obviously missed this during my first successful run.
When it works, it highlights only the items in the module with the selected master part.
When it fails, it highlights all items on the board.

I hit OK and, 15 copies. It worked.
I immediately archived the project.
I went back to the original work, the one with 1 successfully replicated into 2-4.
I clipped all 16 tracks going into the four blocks and failure.
I drug blocks 2-4 aside, deleted all their tracks, putting them back to just parts floating out in space.
Now I am back to one completed block with no tracks connecting it.
Run the plugin and it fails.

I realized that it was highlighting the entire board, no matter what I did.
The "Replicate intersecting tracks" setting has no effect.
The "Remove existing tracks setting" causes it to erase the entire board, if checked.
And I never tried the "Remove Duplicates" setting.
Here is an example.
[image]

The master part Q8, is under the cross hair.
QuadDriver2 (block to the left of master block) is the only block selected in the plugin.
Everything on the board is highlighted and the plugin will fail as shown above.
Notice the parts in the lower left corner, these belong to the common power supply.

I have now moved that only successful attempt that had 16 good copies of the block to my mainstream folder and haven't looked back.
I am close to complete now and wanted a break so I turned here to see what was up.

I hope my story might be helpful.

Thank You

MitjaNemec, 2022-05-21:
Yeah, the post somewhat clears up what was happening, thanks for figuring this out. But I still have no clue why this was happening. In order for me to figure this out I'd need either:

1. you project
2. `replicate_layout.log` file from the time that you observed this. In all likelihood I'd then change some plugin code and ask you to run the plugin again and share the .log file.

Without one of these it is unlikely for me to resolve this

eewiz, 2022-05-22:
Hello,
I can give you the project files minus the schematics if that would help.
I have tested the files with the schematics stripped and the erroneous behavior remains.
If yes, please tell me quickly how to upload the zip using github.

All for now.

MitjaNemec, 2022-05-22:
This would help. You just drag it in the text area when submitting a response

eewiz, 2022-05-22:
Here's everything minus the schematics and the log file.
[ExampleProject.zip](https://github.com/MitjaNemec/ReplicateLayout/files/8749911/ExampleProject.zip)

All for now.

MitjaNemec, 2022-05-23:
Hmm, I can not recreate the issue with your example. If you can, I'd really appreciate if you can describe the steps (as verbosely as you can) with which you can recreate the issue. Thanks for the persistence

eewiz, 2022-05-23:
I installed a brand new copy of KiCAD 6.04 on my laptop.
It never had KiCAD installed before now.
I installed your plugin through KiCAD Manager. The only one installed.
I opened ExampleProject\YM.kicad_pro.
From KiCAD I opened YM.kicad_pcb.
I zoomed in and selected the master component.
[image]

I selected the plugin.
[image]

The plugin started.
[image]

I unchecked Zones.
[image]

I hit the OK button.
[image]

Two different computers, same result.
Q2 is not in the QuadDriver1-16 hierarchy.
Also, notice that the log just stops with no error message.

All for now.

MitjaNemec, 2022-05-27:
First of all thanks for the testing and the report.

But I can not recreate the issue. I am on Kubuntu 20.04 though, and you are on windows I belive. It might be a wx issue as when I run the plugin, the "Source hierarchy level" is selected by default. It'll take a while until I get access to windows machine and test it there. If you've got time, can you rerun the test but select the "Quad_driver.kicad_sch" in the "Source hirarchy level" ListBox?

eewiz, 2022-05-30:
That Did it.
As soon as I selected the "Quad_driver.kicad_sch" in the "Source hirarchy level" ListBox the hierarchical sheets appeared in the "Destination Sheets" ListBox.
I deselected "Replicate Zones" and hit OK.
It worked perfectly.

So, windows users must first select (click on) the hierarchical schematic shown in the "Source hirarchy level" ListBox before any other maneuver.

Thank You

MitjaNemec, 2022-05-30:
Thanks for the testing. I'll add some logic to select bottom most layer at the start

MitjaNemec, 2022-08-06:
This should be fixed in 1.1.3 and later

closed 2022-08-06 by MitjaNemec