MitjaNemec/ReplicateLayout #15: After the undo operation, the plugin tracks and vias are not deleted

HaydenHu, 2022-06-22:
[image]

MitjaNemec, 2022-06-22:
Hi, thanks for the report.

On which KiCad version are you running the plugin? Which version of the plugin are you running? You can attach `replicate_layout.log` which should be in the project folder.

If you save the board after the undo operation, close the KiCad and open it again, are the tracks and vias still there. It might be the case that only graphics is not refreshed. In any case I doubt I'll be able to do much in the short run

HaydenHu, 2022-06-22:
Kicad6.0.6
replicate_layout1.1.2
[replicate_layout.log](https://github.com/MitjaNemec/ReplicateLayout/files/8956398/replicate_layout.log)

InstantMuffin, 2022-10-21:
> If you save the board after the undo operation, close the KiCad and open it again, are the tracks and vias still there. It might be the case that only graphics is not refreshed. In any case I doubt I'll be able to do much in the short run

Hello,

I, too, stumbled upon this problem. Furthermore you can actually not select the tracks and vias. This is solved by saving the project and restarting KiCad.

gunterkoenigsmann, 2022-11-04:
````
Application: KiCad

Version: 6.0.7+dfsg-1build1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.85.0 OpenSSL/3.0.5 zlib/1.2.11 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.3 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.49.0 librtmp/2.3

Platform: Linux 5.19.0-23-generic x86_64, 64 bit, Little endian, wxGTK, KDE, x11

Build Info:
	Date: Aug 10 2022 17:38:39
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.84.0
	ngspice: 37
	Compiler: GCC 12.1.0 with C++ ABI 1017

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
````
After using your excellent extension the first time the undo function actually worked. But then I changed another bit of the layout, told the extension to delete all lines and to replicate the blocks anew - and this was when the undo functionality broke.

MitjaNemec, 2022-11-04:
Thanks for additional info. The main problem with this issue is that I can not recreate it. I'll admit the plugin might be doing something funky which might not be expected or supported by KiCad backend, but it is hard for me to find out what is causing this. It might be my plugin or it might be a bug in KiCad.

It might be the "Remove existing ..." feature that is causing this or it might be "Delete duplicates" causing this. It might be the case that when running the plugin multiple times in one session, so that plugin does not/fails to close properly therefore tripping the [KiCad's builtin undo functionality](https://forum.kicad.info/t/force-save-for-undo-in-python/38329/6?u=mitjan)

MitjaNemec, 2022-11-04:
One thing that I can do is to improve the logging. Specifically to store all the logs and not just the latest one. Having the history of the logs available might offer an more insight what is going on. But I am reluctant to change this as this would either lay waste to the users project folder with numerous logging files or make logging file ridiculously long. I'll have to think about this.

hildogjr, 2023-11-20:
Same on v7.99

MitjaNemec, 2023-11-21:
@hildogjr , do you have a test project or a project that you could share for me to test this. Does this also depend on specific plugin settings? Can you document one case of this as verbosely as you can?

MitjaNemec, 2023-11-21:
Also, can you list which additional plugins you have installed, and also listing which modules you have installed in KiCad's python (execute `help('modules')` in KiCad's pcb editor python console)