MitjaNemec/ReplicateLayout #20: plugin doesn't start

robert-binarylabs, 2022-09-22:
trying to load the plugin results in the following error:

> TypeError("namedtuple() got an unexpected keyword argument 'defaults'",)

using plugin v1.2.1 installed via plugin manager
this is my kicad version: 6.0.4-6f826c9f35\~116\~ubuntu18.04.1

MitjaNemec, 2022-09-23:
Thanks for reporting this. Can you look:

1. into the plugin location folder (should be in folder pointed by KICAD6_3RD_PARTY variable) for `replicate_layout_error.log`
2. into the project folder for `replicate_layout.log`

If you find any file, can you please attach them here.

robert-binarylabs, 2022-09-27:
1. this is the content of `replicate_layout_error.log` in the 3rd party folder:
`TypeError("namedtuple() got an unexpected keyword argument 'defaults'",)`
2. this file has only logs regarding a succesful replication from a year ago.

i think the problems originates from a broken python installation. i can't start the scripting console in pcbnew. will investigate further and report back.

robert-binarylabs, 2022-09-27:
yes, this is the only content in the file.

I'm using Python 3.7.14.

Oh, i didn't know that 18.04 isn't supported anymore. I guess it's finally time to do an dist-ugrade then.
I consider this issue closed.

MitjaNemec, 2022-09-27:
Thanks for reporting though, I'll try to make the .log file more verbose

closed 2022-09-27 by MitjaNemec

MitjaNemec, 2022-09-27:
I've looked into this a bit. I am using the defaults argument with namedtuple on line 35 in `replicate_layout.py`. While this should not cause an issue with python 3.7 and up, I am not really certain, that KiCad is using python 3.7 in your case. I suspect that you've got older python versions available on your system and KiCad is using some version that is older than 3.7

Anyhow, you might change the mentioned line as in https://github.com/ncarrier/discus/issues/1#issue-700635407

robert-binarylabs, 2022-09-27:
wow, that actually worked! thanks a lot.

MitjaNemec, 2022-09-27:
I am glad to hear this. Can you now attach the `replicate_layout.log` it should have info on which python version is actually used. Thanks

MitjaNemec, 2022-09-27:
Ok this confirms the source of the bug. Why KiCad uses python 3.6 instead of 3.7 that also exists on your system. It'll remain a mystery.

Thanks for the info.