MitjaNemec/ReplicateLayout #21: Problems with symbols parts

krivospitsky, 2022-10-09:
I have 8 hieratical sheets, contains some uniqe symbols and one 8-parts symbol (J1)
[image]
[image]
 in this case plugin can't find parts of this multy-part symbol to replicate
[image]

MitjaNemec, 2022-10-10:
Whoa. Thanks for reporting this.

I have a few follow up questions:
1. So you have 8 hierarchical pages all pointing to the same schematics file?
2. Did you annotate the schematics manually?
3. Is use of this kind of symbol in multiple hierarchy schematics supported in 6.0.x?

krivospitsky, 2022-10-10:
1. yes
2. initaly KiCAD assign J1A, J2A, etc to connectors at pages. i changed it to j1am j1b, etc
3. In my situation i creatred some symbols by myself, but i tried to reproduce bug on standard symbols from KiCAD 6.

you can check this small example https://github.com/krivospitsky/demo