MitjaNemec/ReplicateLayout #22: AttributeError: 'NoneType' object has no attribute 'ref'

fergbrain, 2022-10-10:
I'm getting the following error when I try to replicate a layout:
```
Traceback (most recent call last):

File "/Users/REDACTED/Documents/KiCad/6.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/action_replicate_layout.py", line 359, in Run
logger.info(f'source anchor footprint is {repr(src_anchor_fp.ref)}\n'

AttributeError: 'NoneType' object has no attribute 'ref'
```

**Version info:**
ReplicateLayout Version: 1.2.1

Application: KiCad PCB Editor

Version: (6.0.8-1)-1, release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.79.1 SecureTransport (LibreSSL/3.3.6) zlib/1.2.11 nghttp2/1.45.1

Platform: macOS Version 12.5 (Build 21G72), 64 bit, Little endian, wxMac

Build Info:
	Date: Sep 28 2022 15:09:52
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.3
	Curl: 7.64.1
	ngspice: 37
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

MitjaNemec, 2022-10-11:
Hi, Thanks for report. Can you attach `replicate_layout.log` file which should be in the project folder. You should look at it first to remove private information.

fergbrain, 2022-10-11:
Well, this is weird. I went to go try it again this morning and now it works just fine. I must have been doing something wrong 🤦‍♂️

closed 2022-10-11 by fergbrain

MitjaNemec, 2022-10-11:
Well, I've been looking at the code and I could not see what went wrong. But then again with bus like these it usually turns out I just lack imagination. So if it happens again, make sure to store the .log file.