MitjaNemec/ReplicateLayout #23: Copy all elements that are part of the group

InstantMuffin, 2022-10-21:
Hello,

First off, excellent plugin. I have just started using this, and it saved me a lot of trouble so far.
However I am having a little nitpick that makes using this a bit tedious as I still have to do manual corrections.
It often happens that some components are not being copied. This is already noticeable in the plugin's submenu by looking at the preselection of the components in the PCB Editor. Any number of combinations of the options does not allow me to get the very last missing pieces selected.

I feel like this has to do with me creating that subcircuitry with some small unconnected ends. If you like I can recreate this later on and post screenshots and logs, but right now I am honestly too tired, as I made little mistakes in my design and had to redo the replication and the later correction with the tedious selections over and over again and been on this for hours. :(
Rather than it being a bug, I think it would be nice to have an option that just replicates every element that is part of the group of the selected component. That way, and this is how I thought it would work initially, everything that I placed in that group, tracks, vias, silkscreen texts, etc. will be selected automagically.

I'm using KiCad 6.0.8-f2edbf62ab~116~ubuntu22.04.1, release build, and the plugin version is 1.2.1.

Thank you so much for your work and your continued efforts!

InstantMuffin, 2022-10-21:
On second thought, might be what I am experiencing related to what is being referred to as "missing tracks" in #13?

MitjaNemec, 2022-11-04:
Thanks for reporting this.
> It often happens that some components are not being copied. This is already noticeable in the plugin's submenu by looking at the preselection of the components in the PCB Editor. Any number of combinations of the options does not allow me to get the very last missing pieces selected.
I'd need more info on this. For starters I'd need a `replicate_layout.log` from the time that you ran the plugin, and your actual project that you are working on, if you can share it. 

As for
> an option that just replicates every element that is part of the group of the selected component. 
this is currently to big of a problem to solve. Mainly as KiCad's schematics hierarchy feature solve all the connectivity checking that would in this case have to be implemented within the plugin. You can read more on this at https://github.com/MitjaNemec/Kicad_action_plugins/issues/123

InstantMuffin, 2022-11-04:
Hello,

Thank you for your reply!
I read the discussion in #123 and I see that grouping is a bit more complicated than it seems to the user.
I'll work on a very bare sample project to reproduce this and upload here.

InstantMuffin, 2022-11-04:
Here you can see some vias and some silkscreen texts will be missing. They are all part of a group.
[image]

I applied the changes/did the replication, and zipped and uploaded the project so that you have the log and the result.

[replicatelayouttest.zip](https://github.com/MitjaNemec/ReplicateLayout/files/9942302/replicatelayouttest.zip)

MitjaNemec, 2022-11-05:
From the picture it seems that the plugin only grabs item that intersects bounding box created by footprints from the source hierarchical page. It disregards other items which are also members of the group that the source anchor footprint belongs to.

This could be solved by adding another checkbox "Replicate also the items belonging to the group that the source anchor footprints belongs to". But:
1. The UI is becoming more and more cluttered and I have some reservations about adding another checkbox
2. The proposed text is just to long

So I am looking for ideas how to implement this properly within the UI as this is obviously not really my forte. I've noticed this in numerous issues/pull request during the years. The last one was from yesterday ([Save/Restore Layout #8](https://github.com/MitjaNemec/SaveRestoreLayout/issues/8))

MitjaNemec, 2022-11-05:
Thanks for the test case, I'll look into it and if it is what I am assuming, I'll work on the solution

MitjaNemec, 2022-11-05:
I have one more question. Should the plugin replicate also tracks and vias which are otherwise completely outside the bounding box, but are part of the group? This seems a dangerous option

InstantMuffin, 2022-11-05:
Hello,

Maybe a little background to myself: I have only been using KiCad for about half a year now, and only build around 10 projects with it. I have no other experience with any other electronics design software. So I don't know if there is a better way or workflow to do these things.

But for me, creating component groups seems to be naturally done using the following steps:
1) Working on the hierarchical sheet in the schematic editor
2) Duplicating it
3) Updating the pcb with the new components
4) Wiring up one component group completely (placements, tracks, vias, silkscreen texts, etc.) without connecting it to anything outside of the hierarchical sheet.
5) Then using "something" (and that is where your plugin has been recommended to me) to apply that exact pcb design to the other hierarchical sheet duplicates.

For me it makes perfect sense to group the entire component group created in step 4. I'll be moving that group in its entirety, and all the pieces belong together logically and logistically.
I didn't only create a group for myself and the handling convenience, but I naturally expected the plugin to actually need the components to be part of a group, as the hierarchical sheet itself does not account for anything other the bare components.

I can't speak for anyone else, and I would love to have some input on this, someone more experienced maybe suggesting a better way of doing this than I have done so far. But this is both my way of doing this as well as my journey as a newbie to solving this particular problem (which might be interesting from a user experience standpoint).

I would absolutely love what you're suggesting and what I have been requesting to be an option and therefore addition to your plugin.

MitjaNemec, 2022-11-06:
Thanks for the background. As far as the plugin features it always comes down to getting an idea. And I obviously deal with different layout as I don't seem to recall that I had any items outside the bounding box (formed by footprints) that I wanted to replicate. So I am always glad to get UX pointers from other people.

As for the technical issues with this feature I see three cases:
1. it should be quite easy to implement replication of text and drawings outside the bounding box.
2. Zones, tracks and vias are a bit of a tougher nut to crack due to connectivity. E.g. if the zone/track/via in question is connected to a net that is within the hierarchical sheet, then the replicated zone/track/via can be connected to appropriate net. But if zone/track/via is connected to a net that does not exist within the hierarchical sheet, then the plugin can not figure where to which net should connect replicated item.
3. Finally replication of footprints which are within the group but outside the hierarchical sheet is impossible without implementing half of Kicad backend within the plugin. So this will not bi implemented

The only question the is how to implement 2 and I can see only one sensible way of doing it. In this case the plugin would replicate only zones/tracks/vias which are connected to hierarchical sheet internal nets. All other items would not get replicated. 

I'll look into this

MitjaNemec, 2022-11-16:
Well, implementation of this feature pushed me to do some refactoring, which is always good so that the codebase gets some improvement. Anyhow, the fix for your issue is available in version 1.2.3, which you can download under Releases.

I'd appreciate if you can test it a bit, before I push it to KiCad main repository