MitjaNemec/ReplicateLayout #27: Cannot see all hierarchical sheets when trying to replicate a circuit with nested hierarchical sheets

afvincent, 2022-11-21:
Hi,

First thank you for the plugin :). I am struggling with something that might be more a misunderstanding by myself than a real issue but I did not know a more appropriate place to check if that is indeed the case. I apologize if there is actually a better place and please feel free to redirect me.

I use KiCad 6.0.8 and ReplicateLayout 1.2.2 (on Fedora 35).

As far as I understand the description ReplicateLayout on the front page of its GitHub repository, it should be able to replicate partial layouts of nested hierarchical sheet, should it not?

In particular, I am trying to use ReplicateLayout in the following situation:

Root
  |
  |-> Lvl0-Inst0
  |       |
  |       |-> Lvl1-Inst0     
  |       |-> Lvl1-Inst1     
  |       |-> Lvl1-Inst2     
  |
  |-> Lvl0-Inst1
  |       |
  |       |-> Lvl1-Inst0     
  |       |-> Lvl1-Inst1     
  |       |-> Lvl1-Inst2     

where the bottom level (`Lvl1-Inst*`) contains the subcircuit that I would like to replicate (e.g., 2 × 3 times here).

Here are eeschema screenshots of the content of the different hierarchy levels: 

Root:
[image]

Lvl0:
[image]

Lvl1:
[image]

I built that hierarchy through the following procedure:
1. Create `Lvl0-Inst0` in `Root`.
2. Create `Lvl1-Inst0` in `Lvl0-Inst0`.
3. Inside `Lvl1-Inst0`, draw the schematics of the subcircuit to be replicated and assign footprints.
4. Go up to `Lvl0-Inst0` and copy-paste `Lvl1-Inst0` in order to get `Lvl1-Inst1` and `Lvl1-Inst2`.
5. Go up to `Root` and copy-paste `Lvl0-Inst0` in order to get `Lvl0-Inst1`.
6.  Add a connector and a decoupling capacitor directly at the `Root` level.
7. Automatically assign references to the whole schematic.
Thus all the `Lvl1-Inst*` sheets should be exact copies of the first one created (except for the references).

Then I go to the PCB editor and place and route one of the 6 subcircuits from `Lvl1-Inst*`.

Finally, when I select one of the routed footprints and try to run ReplicateLayout, I get the following dialog window, where I can only see only 3 different hierarchical sheets (including the already routed one) instead of the 6 ones that I would expect:
[image]

Am I doing something wrong or did I misunderstand the description on the front page of the plugin repository?

PS: Just in case, “Fichier” means “File” in French.

MitjaNemec, 2022-11-21:
Thanks for reporting. You should definitely see two hierarchical levels in the plugin dialog. 

Your issue might be due to a bug in 6.0.8, so I'd first like you to update to 6.0.9 and make sure you update the layout from schematics, so that underlying data structures get fixed.

If this does not help, then please attach the 'replicate_layout.log' file which should be in the project folder.

afvincent, 2022-11-22:
I just installed KiCad 6.0.9 an tried again (after updating the PCB from the schematics twice, just to be sure). I get almost the same result as with version 6.0.8, except that this time it is the sheets that contain `D2` and `D3` (instead of `D5` and `D6` previously) that appear in the plugin dialog:

[image]

Here is the content of the log file output by KiCad 6.0.9:
```
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 312:Plugin executed on: 'linux'
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 313:Plugin executed with python version: '3.10.7 (main, Sep  7 2022, 00:00:00) [GCC 11.3.1 20220421 (Red Hat 11.3.1-2)]'
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 314:KiCad build version: 6.0.9-1.fc35
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 315:Plugin version: 1.2.2
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 316:Frame repr: <wx._core.Frame object at 0x7f14f9b0e320>
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 340:Preparing replicator with D1 as a reference
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.replicate_layout 106:getting a list of all footprints on board
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 364:source anchor footprint is 'D1'
Located on: ['Lvl1-Inst0']
With filenames: ['Lvl1.kicad_sch']
With sheet_id:['Lvl1-Inst0']
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 371:Corresponding footprints are 
[('D4', ['Lvl1-Inst0']), ('D2', ['Lvl1-Inst1']), ('D3', ['Lvl1-Inst2']), ('D6', ['Lvl1-Inst2']), ('D5', ['Lvl1-Inst1']), ('D1', ['Lvl1-Inst0'])]
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 383:Showing dialog
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.replicate_layout 344:constructing a list of sheets suitable for replication on level:'Lvl1-Inst0', file:'Lvl1.kicad_sch'
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.replicate_layout 381:suitable sheets are:[['Lvl1-Inst1'], ['Lvl1-Inst2']]
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.replicate_layout 1112:Level selected: ['Lvl1-Inst0']
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 389:Toolbar position: wx.Point(2526, 130)
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 395:Dialog position: wx.Point(2162, 130)
11-22 07:58:17 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 241:User canceled the dialog
```

As far as as understand what is going on, it looks like the plugin is successful at identifying all the anchor footprints that I would like to use (`D1`, `D2`, `D3`, `D4`, `D5` and `D6`):
```
11-22 07:56:32 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 371:Corresponding footprints are 
[('D4', ['Lvl1-Inst0']), ('D2', ['Lvl1-Inst1']), ('D3', ['Lvl1-Inst2']), ('D6', ['Lvl1-Inst2']), ('D5', ['Lvl1-Inst1']), ('D1', ['Lvl1-Inst0'])]
```
but it seems to be struggling with the fact that each bottom level sheet (`Lvl1-Inst*`) actually appears twice.

afvincent, 2022-11-22:
Here is a ZIP archive that contains the files for the small KiCad project shown above: 

[test_project_for_issue_with_replicatelayout.zip](https://github.com/MitjaNemec/ReplicateLayout/files/10065655/test_project_for_issue_with_replicatelayout.zip)

MitjaNemec, 2022-11-22:
OK, I've found a bug but it'll be a while before I fix it. The issue is that you don't have any symbols/footprints on `Lvl0` and then the plugin fails to derive the project hierarchy properly. So in the meanwhile you can place a dummy symbol with footprint on this level, so that you can run the plugin and then you can delete it.

afvincent, 2022-11-22:
Thank you very much for the workaround!

I can confirm that adding a dummy symbol with a footprint in the hierarchical level(s) without any symbol (other than hierarchical sheets) allows to successfully run the ReplicateLayout plugin *as long as the dummy footprint(s) are present in the layout, when one runs the plugin.* Once it is done, one can then simply remove the dummy footprint(s).

As far as I am concerned, I can definitively live with that until you find a way to fix the bug (if you find the time to do that). I guess that my use case example is a bit of a corner case anyway (it was actually more of a highly simplified test case for a larger system, where I expect to have quite a few levels of hierarchy).

MitjaNemec, 2022-11-23:
Thanks for confirmation. As a principle I try to make my plugins as hassle free as I can. And that includes proper handling of corner cases.

If I can't solve the issue properly at least the plugin should report back the issue and recommend the workaround. But I'll first try to solve the issue. This makes my life easier as it reduces the number of issues raised.

MitjaNemec, 2022-12-01:
It took some time, as I pushed the update also to other two plugins (Place Footprints and Save/restore layout) and I had some issue with the latter one. The fix is commited and updated versions of the plugins should be available soon in Kicads PCM

closed 2022-12-01 by MitjaNemec