MitjaNemec/ReplicateLayout #29: IndexError: list index out of range

manotam, 2022-12-12:
First of all thank you very much for the plugin. It helps a ton!

I installed "replicate layout" via the Plugin and Content Manager as suggested.

I select a component and hit the plugin button and get the following message 

[image]

It seems to work for others. I am using windows 10

Thanks!

MitjaNemec, 2022-12-13:
Thanks for reporting this. From the error message allone, I have a hard time figuring out why this happens from the error message alone. Can you review the `replicate_layout.log` file for any personal information and then attach it here ? The file should be in the project folder. Can you maybe share the project?

manotam, 2022-12-13:
Hey!

Thanks fo the speedy response. I uploaded a test project with the same error into a repo:

https://github.com/manotam/KicadTest

MitjaNemec, 2022-12-13:
Thanks for the project

The main issue is that the plugin can replicate only (and only) parts of the layout which derive from the same hierarchical sheet. In the attached project this means that each RC circuit should be an instance of the same circuit described by one hierarchical sheet file. I am attaching refactored project.

BTW, You should update KiCad to 6.0.9+ and update the layout from schematics before running the plugin, as 6.0.8 has a bug which is shows when running my plugins.

And finally thanks again for reporting and sending a project. This cas was never tested and the plugin should not crash handling it. It should issue a proper error message. I'll get on to it.

[manotam_issue.zip](https://github.com/MitjaNemec/ReplicateLayout/files/10220271/manotam_issue.zip)

manotam, 2022-12-13:
Ah,
now I see what you mean. I was not aware of this functionality in Kicad. This issue might have derived from my lack of knowledge, but good to hear that it was helpful regardless!

Thank you for your help! Also updated my Kicad :P

closed 2022-12-21 by MitjaNemec