MitjaNemec/ReplicateLayout #31: FileNotFoundError: [Errno 2] No such file or directory: '.kicad_sch'

ruR2D2, 2022-12-19:
20.12.2022
[image]

12-20 02:01:20 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 326:Plugin executed on: 'win32'
12-20 02:01:20 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 327:Plugin executed with python version: '3.9.14 (main, Sep 19 2022, 05:11:57) [MSC v.1932 64 bit (AMD64)]'
12-20 02:01:20 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 328:KiCad build version: (6.0.9)
12-20 02:01:20 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 329:Plugin version: 1.2.6
12-20 02:01:20 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 330:Frame repr: <wx._core.Frame object at 0x0000028D14E1C430>
12-20 02:01:20 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 354:Preparing replicator with J87 as a reference
12-20 02:01:20 com_github_MitjaNemec_ReplicateLayout.replicate_layout 110:getting a list of all footprints on board
12-20 02:01:20 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 369:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "C:\Users\anton\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\action_replicate_layout.py", line 358, in Run
    replicator = Replicator(board, src_anchor_fp_reference)
  File "C:\Users\anton\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 153, in __init__
    self.parse_schematic_files(self.sch_filename, schematic_found)
  File "C:\Users\anton\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 198, in parse_schematic_files
    self.parse_schematic_files(sheetfile, dict_of_sheets)
  File "C:\Users\anton\Documents\KiCad\6.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 179, in parse_schematic_files
    with open(filename) as f:
FileNotFoundError: [Errno 2] No such file or directory: '.kicad_sch'

cyntem, 2022-12-20:
Hi, I think you have the same bug with python lstrip that's I had.
```
>>> "act.kicad_sch".lstrip("(property \"Sheet name\"") 
'ct.kicad_sch'

```
I did a fork https://github.com/cyntem/ReplicateLayout and fixed it there, I did pull request to current repository.

MitjaNemec, 2022-12-21:
Thanks for reporting. That is what I get if I test new versions only and only on one target (Linux).

There is already PR #30  which might address this issue, I'll just have to find the time to test this on both platforms. Please be patient.

MitjaNemec, 2022-12-23:
Should be fixed in 1.2.7

closed 2022-12-23 by MitjaNemec