MitjaNemec/ReplicateLayout #33: UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 45029: ordinal not in range(128)

W-M-D, 2023-01-04:
```01-04 08:37:16 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 326:Plugin executed on: 'linux'
01-04 08:37:16 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 327:Plugin executed with python version: '3.10.9 (main, Dec 19 2022, 17:35:49) [GCC 12.2.0]'
01-04 08:37:16 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 328:KiCad build version: (6.99.0-3191-gcfdf6c097d)
01-04 08:37:16 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 329:Plugin version: 1.2.7
01-04 08:37:16 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 330:Frame repr: <wx._core.Frame object at 0x6cf9fb763d90>
01-04 08:37:16 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 354:Preparing replicator with D1 as a reference
01-04 08:37:16 com_github_MitjaNemec_ReplicateLayout.replicate_layout 110:getting a list of all footprints on board
01-04 08:37:16 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 369:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "/home/matt/.local/share/kicad/6.99/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/action_replicate_layout.py", line 358, in Run
    replicator = Replicator(board, src_anchor_fp_reference)
  File "/home/matt/.local/share/kicad/6.99/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 153, in __init__
    self.parse_schematic_files(self.sch_filename, schematic_found)
  File "/home/matt/.local/share/kicad/6.99/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 180, in parse_schematic_files
    contents = f.read().split("\n")
  File "/usr/lib/python3.10/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 45029: ordinal not in range(128)
```

MitjaNemec, 2023-01-04:
Thanks for reporting.

This is a strange one. I'd assume that utf-88 should be default encoding when reading files.

If you can, can you pleas open python console in pcbnew and type in following
```python
import sys
sys.getfilesystemencoding()
```
and report back the output.

As for the fix, until the new version is published you can try changing the line 179 in file `/home/matt/.local/share/kicad/6.99/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py` to 
```python
        with open(filename, encoding='utf-8') as f:
```

MitjaNemec, 2023-01-04:
Additionally, this should not really happen. Do you have any other plugins installed? And if so, which plugins do you have?
It looks like your locale setting are changed.

commit d319153d (2023-01-05): Add forgoten V5 porting code to schematic parsing code. Enforce utf-8 encoding when reading schematics files.

Fixes #34
Fixes #33
closed 2023-01-05 by MitjaNemec