MitjaNemec/ReplicateLayout #35: Replicated Text Position Error

myamigo, 2023-01-09:
The project in [Replicate_Text_Error.zip](https://github.com/MitjaNemec/ReplicateLayout/files/10375897/Replicate_Text_Error.zip) exposes this issue.

The project consists of 3 copies of a sheet containing 1 connector. The PCB places the seed component (J1) at (100,100) with the drill origin at (150,100).

J2 and J3 are first positioned relative to J1 and then rotated ±120° around the drill origin so we have this as a starting point...
[image]

Next run the replicator using J1 as the seed with all defaults selected...
[image]

... and when finished you will notice that the text for J2 & J3 are repositioned slightly and that the ref text actually overlaps the pads. This triggered an unexpected DRC error, which revealed the problem.

[image]

[image]
[image]
[image]

MitjaNemec, 2023-01-10:
Whoa, it seems I have a numerical resolution problem somewhere. Thanks for being meticulous enough to spot this and especially for following through and reporting it. I really appreciate it.

myamigo, 2023-01-10:
Glad to help! I also discovered that the "Keep upright" property for text in footprints now shows that relative to the side of the display rather than the bottom. I reported this in https://gitlab.com/kicad/code/kicad/-/issues/13460

I am hoping this gets resolved by the KiCad team. If that is valid maybe you could add a comment to raise the flag higher?

closed 2023-01-24 by MitjaNemec

MitjaNemec, 2023-01-24:
Well thanks again for spotting this and reporting with a clear test case. It stayed hidden for a really long time.

Anyhow the release is available for manual (from file) installation at https://github.com/MitjaNemec/ReplicateLayout/releases/tag/1.2.11

It will take a while before it is available via PCM as I'd like to cherry pick this also in V7 compatible version, but I am having some issues.

myamigo, 2023-01-24:
Tested 1.2.11 on my design and it's good. Thank you for handling this one.

commit fcf7038d (2023-01-24): use proper information for center of text and center of footprint when positioning footprint text.

Fixes #35