MitjaNemec/ReplicateLayout #37: KC 7.0 RC with Replicator 2.0.1

myamigo, 2023-01-10:
I tried the latest KC 7.0 RC nightly build with your 2.0.1 replicator and I had issues with the KC PCM finding the URL for your installer but that seemed to be a problem with other plugins as well. I downloaded the 2.0.1 ZIP and installed it manually and got by that issue but found the replicator failed with a familiar error...
[image]

Log here... [replicate_layout.log](https://github.com/MitjaNemec/ReplicateLayout/files/10386239/replicate_layout.log)

I tried applying the fix you offered before ... https://github.com/MitjaNemec/ReplicateLayout/issues/34#issuecomment-1371947498 ... but that didn't help.

If we can get your replicator to work with the KC 7.0 RC then I think I might be OK with going to that.

myamigo, 2023-01-11:
I think I can get by using KC 6.0 and the pending fix to https://github.com/MitjaNemec/ReplicateLayout/issues/35 and just use KC 7.0 when I need to generate outputs so the component text is oriented correctly. This seems like the least risky option.

myamigo, 2023-02-16:
I just installed the stable release of KC 7.0 to try 2.0.3 of the replicator. It failed with the familiar root sheet error...
[image]
... so I tried applying the fix you offered before ... [#34 (comment)](https://github.com/MitjaNemec/ReplicateLayout/issues/34#issuecomment-1371947498) and this time it worked. That looks like a safe update to apply over 2.0.3

myamigo, 2023-02-16:
I also note that the text offset error described in https://github.com/MitjaNemec/ReplicateLayout/issues/35 is still present but I gave up trying to figure out what needs to be done to get that working. I'm sure it'll be straight forward for you though. 

When these are sorted out then I will try working more with KC7 to see if there are any other issues.

MitjaNemec, 2023-02-17:
Thanks for additional info. Currently I am a bit short on time, so the progress is quite slow.

Germwalker, 2023-03-03:
Hello, 

I'm on RC7 also and i have the same problem with the replicator. 
I have 100 channels of the same sheet to place so i'm very very interested in a quick fix.
I'm ready to help ! 

Here is the log : 

```
03-03 10:47:02 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 326:Plugin executed on: 'linux'
03-03 10:47:02 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 327:Plugin executed with python version: '3.9.2 (default, Feb 28 2021, 17:03:44) \n[GCC 10.2.1 20210110]'
03-03 10:47:02 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 328:KiCad build version: (7.0.0-rc1-342-g4eb3a99bb5)
03-03 10:47:02 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 329:Plugin version: 2.0.1
03-03 10:47:02 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 330:Frame repr: <wx._core.Frame object at 0x7ff4da7939d0>
03-03 10:47:02 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 354:Preparing replicator with U13 as a reference
03-03 10:47:02 com_github_MitjaNemec_ReplicateLayout.replicate_layout 110:getting a list of all footprints on board
03-03 10:47:02 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 369:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "~/.local/share/kicad/7.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/action_replicate_layout.py", line 358, in Run
    replicator = Replicator(board, src_anchor_fp_reference)
  File "~/.local/share/kicad/7.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 153, in __init__
    self.parse_schematic_files(self.sch_filename, schematic_found)
  File "~/.local/share/kicad/7.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 199, in parse_schematic_files
    self.parse_schematic_files(sheetfile, dict_of_sheets)
  File "~/.local/share/kicad/7.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 179, in parse_schematic_files
    with open(filename, encoding='utf-8') as f:
FileNotFoundError: [Errno 2] No file or folder of this type: ''

```
It seems to be a filename problem ?

MitjaNemec, 2023-03-03:
@Germwalker  you should update the plugin as I think that the bug has been fixed

MitjaNemec, 2023-04-17:
Thanks for reporting back