MitjaNemec/ReplicateLayout #40: FileNotFoundError: [Errno 2] No such file or directory: ''

tinfever, 2023-02-06:
When trying to use ReplicateLayout with the latest KiCad 7 nightly, I was receiving the error:
FileNotFoundError: [Errno 2] No such file or directory: ''

I did some digging and found that my schematic files were now using the property "Sheetname" and "Sheetfile", while the script expects "Sheet file" and "Sheet name".

To fix this I made the following change (removed the space between the two words in the script):

Before (replicate_layout.py line 192-195): 
```
if "(property \"Sheet name\"" in contents[j]:
    sheetname = contents[j].replace("(property \"Sheet name\"", '').split("(")[0].replace("\"", "").strip()
if "(property \"Sheet file\"" in contents[j]:
    sheetfile = contents[j].replace("(property \"Sheet file\"", '').split("(")[0].replace("\"", "").strip()
```

After (replicate_layout.py line 192-195): 
```
if "(property \"Sheetname\"" in contents[j]:
    sheetname = contents[j].replace("(property \"Sheetname\"", '').split("(")[0].replace("\"", "").strip()
if "(property \"Sheetfile\"" in contents[j]:
    sheetfile = contents[j].replace("(property \"Sheetfile\"", '').split("(")[0].replace("\"", "").strip()
```

This change fixes the issue. I don't know when or why this was changed on KiCad's part. Since I was using this plugin yesterday and it was fine. I did a bunch of hierarchical sheet restructuring today so perhaps this is a KiCad 7 change that only was applied when making and renaming sheets.

More info:

replicate_layout.log:
```
02-06 00:01:15 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 326:Plugin executed on: 'win32'
02-06 00:01:15 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 327:Plugin executed with python version: '3.9.14 (main, Dec 23 2022, 23:36:19) [MSC v.1934 64 bit (AMD64)]'
02-06 00:01:15 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 328:KiCad build version: (7.0.0-rc2-187-g9b9ebebb2d)
02-06 00:01:15 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 329:Plugin version: 2.0.1
02-06 00:01:15 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 330:Frame repr: <wx._core.Frame object at 0x0000019A4A8D20D0>
02-06 00:01:15 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 354:Preparing replicator with J1501 as a reference
02-06 00:01:15 com_github_MitjaNemec_ReplicateLayout.replicate_layout 110:getting a list of all footprints on board
02-06 00:01:15 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 369:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "C:\Users\user\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\action_replicate_layout.py", line 358, in Run
    replicator = Replicator(board, src_anchor_fp_reference)
  File "C:\Users\user\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 153, in __init__
    self.parse_schematic_files(self.sch_filename, schematic_found)
  File "C:\Users\user\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 199, in parse_schematic_files
    self.parse_schematic_files(sheetfile, dict_of_sheets)
  File "C:\Users\user\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 179, in parse_schematic_files
    with open(filename, encoding='utf-8') as f:
FileNotFoundError: [Errno 2] No such file or directory: ''
```

Kicad version info:
```
Application: KiCad PCB Editor x64 on x64

Version: (7.0.0-rc2-187-g9b9ebebb2d), release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.12.1
	HarfBuzz 5.0.1
	FontConfig 2.14.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb  3 2023 06:19:21
	wxWidgets: 3.2.1 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.83.1-DEV
	ngspice: 38
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

MitjaNemec, 2023-02-06:
First, thanks for the report. Bugs get fixed only when they are reported.

Secondly, I was aware of this bug and I have a fix for it, but as I am also fixing a couple of other things and I have a difficult time testing  V7 (I am on Ubuntu, where plugins in V7RC don't work yet) I haven't yet published the fixed version.

As for your observation
>  I don't know when or why this was changed on KiCad's part. Since I was using this plugin yesterday and it was fine. 
I am not certain when did the schematic file format change, but I think it was late in the V6.9 development. So if you started the project with 6.99 the layout might had properties annotated with "Sheet file". These are updated when you "update layout from schematics". So if you did this yesterday it would explain why the plugin failed.

Another possible explanation for this is that I've also updated the plugin recently. Previous versions of the plugin did not parse the schematics to derive the project hierarchy. This was added recently to solve a bug. So if you were running older version of the plugin and then updated to the newer one yesterday, this might also explain why the plugin failed.

ivizot, 2023-02-21:
Similar error with 2.0.4.

[replicate_layout.log](https://github.com/MitjaNemec/ReplicateLayout/files/10790114/replicate_layout.log)

MitjaNemec, 2023-02-21:
> BTW, plugin version says its 2.0.1 in the log, however 2.0.4 is installed.

@ivizot how did you install the plugin?

Moreover 2.0.4 was a V7 compatibility release along with all my other footprints. Now that I've looked deeper I've seen another issue and I hope to release 2.0.5 in a day or so

ivizot, 2023-02-22:
@MitjaNemec I have installed if with KiCAD PCM. Really appreciate your work, let me know if you need any further testing :)

MitjaNemec, 2023-02-22:
Can you try 2.0.5, which was just released?

ivizot, 2023-02-22:
Hey,
the single line in the log:
"ModuleNotFoundError("No module named 'com_github_MitjaNemec_ReplicateLayout.conn_issue_GUI'")"

MitjaNemec, 2023-02-23:
Thanks for reporting this. This was an error on my side. I forgot to add GUI dialog file to the package. I've fixe this, but it'll take a day or so before the fixed release becomes available in PCM. You can download the fixed version from [releases](https://github.com/MitjaNemec/ReplicateLayout/releases).

ivizot, 2023-02-27:
@MitjaNemec yes, it is working now! Thank you so much, your plugins had already saved me tens of hours!!! Could you please also fix save/restore layout, where the errors are just the same? I believe the fix would be mostly identical?..

MitjaNemec, 2023-02-27:
@ivizot Fixes should already be in Save/Restore layout plugin (2.0.6)