MitjaNemec/ReplicateLayout #41: [feature request] place replicated items in groups

claudiolorini, 2023-02-09:
Hello Mitja, 
do you think that it could be possible to place automatically the replicated items in groups?
it could help the placement of the replicas, since being a group they would select and move all together.
For example, PADS (mentor / siemens) works with this logic, the replicas are grouped by default.

MitjaNemec, 2023-02-09:
Yeah, I have been discussing this previously but there are some questions that should be resolved.
1. How should the group be named?
2. What to do if the group already exists?
3. What to do if some of the destination footprints are already members of some group? Currently footprints cannot be members of more than one group.

claudiolorini, 2023-02-09:
i'm currently testing with 7.0.0-rc2 (ubuntu 22.10)

1. from the PCB editor it seems that you can group stuff without naming it (if you select a group in the status bar you get: <unnamed>), i don't know if the python interface needs a mandatory name but some header plus the 'sheet-name' seems a reasonable candidate. I haven't found any other place in the PCB editor UI where a group is referenced by name; even the 'properties manager' shows only 'group',  so up to now, by the user point of view, group names doesn't seem to be particularly relevant. 

2. you mean if the group name already exists? maybe some kind of randomization? 

3.  is the trickiest. PADS in this case throws you an error log and does not proceed in the replica of the group with the 'missing' item, i think that is a reasonable behavior. 
the alternative is to break the existing group, and this in my opinion is a big no, as a user it would piss me off.

MitjaNemec, 2023-02-11:
3 is especially problematic, as it will make  reruning the plugin a PITA.

Additionally, adding this feature will/would add a new layer of complexity to the plugin code. Which is something I'd like to avoid.

And there is a workaround possible. Just place destination anchor footprints far appart so you can select and group replicated parts manually.

So this feature is currently on standby.

MitjaNemec, 2023-06-05:
This was implemented by @SiTLOOMS

MitjaNemec, 2023-06-05:
But it is quite simple in implementation and we'll see how it behaves when other group are already present