MitjaNemec/ReplicateLayout #42: ModuleNotFoundError("No module named 'com_github_MitjaNemec_ReplicateLayout.conn_issue_GUI'")

summoningdark, 2023-02-22:
It appears that the command to build this module is missing from "make_a_package.sh"

Lines 8-10:
```
    # refresh the GUI design
    wxformbuilder -g replicate_layout_GUI.fbp
    wxformbuilder -g error_dialog_GUI.fbp
```

probably needs a line:
`wxformbuilder -g conn_issue_dialog_GUI.fbp`

I can't test, since I am on my work computer, and they won't let me have a linux laptop :( but the error shows up when trying to load the plugin in pcbEditor.

I'm using plugin version 2.0.5 and KiCad version 7 on windows 10.

Aylo6061, 2023-02-23:
+1 I am getting the same error in the error log with the 1.3.0 release in kicad 6.10. It seems like the file conn_issue_GUI.py file was added to the gitignore- as a result when installed via package manager I get this same error.  the previous 1.2.x release works fine!

Shackmeister, 2023-02-23:
+1 I get same error on V7.0

MitjaNemec, 2023-02-23:
Thanks for reporting this. This was an error on my side. I forgot to add GUI dialog file to the package. I've fixe this, but it'll take a day or so before the fixed release becomes available in PCM. You can download the fixed version from [releases](https://github.com/MitjaNemec/ReplicateLayout/releases).

summoningdark, 2023-02-23:
That fixed it for me. Thanks for this plugin, it is a lifesaver on large boards.

closed 2023-02-23 by summoningdark