MitjaNemec/ReplicateLayout #46: AttributeError: module 'pcbnew' has no attribute 'BOX2I'

gavinunderdown, 2023-04-06:
This is my first time using the plugin, so please feel free to tell me I'm an idiot and using it wrong. 

Log file: 

`cat replicate_layout.log 
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 359:Plugin executed on: 'linux'
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 360:Plugin executed with python version: '3.9.2 (default, Feb 28 2021, 17:03:44) \n[GCC 10.2.1 20210110]'
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 361:KiCad build version: 6.0.11+dfsg-1~bpo11+1
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 362:Plugin version: 2.0.10
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 363:Frame repr: <wx._core.Frame object at 0x7f65354c0160>
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 387:Preparing replicator with Q1 as a reference
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 113:getting a list of all footprints on board
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.replicate_layout 132:Footprint REF** does not have Sheetfile property, it will not be replicated. Most likely it is only in layout
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 435:source anchor footprint is 'Q1'
Located on: ['Driver2', 'Driver Subcircuit 1']
With filenames: ['/home/gavinu/Documents/Research/nozzle-and-driver/hardware/kicad/driver_full.kicad_sch', '/home/gavinu/Documents/Research/nozzle-and-driver/hardware/kicad/driver_subcircuit.kicad_sch']
With sheet_id:['Driver2', 'Driver Subcircuit 1']
04-06 16:57:07 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 442:Corresponding footprints are 
[('Q18', ['Driver6', 'Driver Subcircuit 1']), ('Q15', ['Driver1', 'Driver Subcircuit 1']), ('Q12', ['Driver5', 'Driver Subcircuit 1']), ('Q9', ['Driver4', 'Driver Subcircuit 1']), ('Q4', ['Driver3', 'Driver Subcircuit 1']), ('Q1', ['Driver2', 'Driver Subcircuit 1'])]
[... cut ...]
  File "/home/gavinu/.local/share/kicad/6.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 1350, in highlight_set_level
    fps_bb = self.get_footprints_bounding_box(src_fps)
  File "/home/gavinu/.local/share/kicad/6.0/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 514, in get_footprints_bounding_box
    bounding_box = pcbnew.BOX2I(position, size)
AttributeError: module 'pcbnew' has no attribute 'BOX2I'
`

If needed I can upload a copy of the PCB file as well. 

Is this a bug or does the problem exist between the keyboard and the chair?

gavinunderdown, 2023-04-06:
[kicad.zip](https://github.com/MitjaNemec/ReplicateLayout/files/11174703/kicad.zip)

Full project file

MitjaNemec, 2023-04-07:
You are using wrong version of the plugin. Plugin releases 1.x.y are compatible with KiCad 6.0.x, while plugin releases 2.x.y are compatible with KiCad 7.0.x.

How did you manage to install this version? This should not be possible as PCM should offer you only version 1.3.5.