MitjaNemec/ReplicateLayout #48: Not replicating single ended tracks

Germwalker, 2023-04-17:
Hello, i'm working on a fairly large board with +100 channels.

Each channel is implemented inside it's own square space :
[image]

If i try to duplicate this system (it's a sheet of sheets), with this setup : 
[image]

I have this result on others channels :
[image]

I would like to copy single ended tracks & vias as a working base for the auto router.

How can i duplicate those items ?

MitjaNemec, 2023-04-17:
In the source layout place complete layout in one group (including tracks and vias) and then check the "include group items" checkbox

Germwalker, 2023-04-17:
I have made a group :
[image]

Entered the group, selected one component then replicated with this config :
[image]

As before i have some warnings (maybe it is the issue ?) :
[image]

Then i have the same replicated pattern : 
[image]

Germwalker, 2023-04-17:
Note : i'm the last version of the plugin with kicad 7.0.0-rc1

Germwalker, 2023-04-17:
Here is the replicate_layout.log
[replicate_layout.log](https://github.com/MitjaNemec/ReplicateLayout/files/11250608/replicate_layout.log)

I will try to make an example project showing the issue.

I could temporary use copy / paste like on altium, but if the copy the whole tracks & vias and paste it over others channels, nets are not reassigning. I could remember it was possible on previous versions, maybe you have an advice for me to use this temporary fix ?

Germwalker, 2023-04-17:
Hello, here is the log, same result
[Uploading replicate_layout.log…]()

MitjaNemec, 2023-06-05:
Thanks for sharing. The [2.1.1](https://github.com/MitjaNemec/ReplicateLayout/releases/tag/2.1.1) should handle this. Can you test if?