MitjaNemec/ReplicateLayout #49: Does not replicate zones or tracks on KiCad 7.0.2

Miceuz, 2023-04-18:
My KiCad version:
```
Application: KiCad x86_64 on x86_64

Version: 7.0.2-6a45011f42~172~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.10.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux Mint 20.3, 64 bit, Little endian, wxGTK, cinnamon, x11

Build Info:
	Date: Apr 17 2023 07:58:01
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_SPICE=ON
```

I have pinpointed the problem to a `Duplicate` call - it returns a `BOARD_ITEM` that does not have `SetNetCode` method, not a `PCB_TRACK` as expected.

[replicate_layout.log](https://github.com/MitjaNemec/ReplicateLayout/files/11259256/replicate_layout.log)

Miceuz, 2023-04-18:
Altering the duplication to cast - `new_track = track.Duplicate().Cast()` makes it work.

MitjaNemec, 2023-04-18:
Thanks for reporting and a solution. This is probably due to https://gitlab.com/kicad/code/kicad/-/issues/14460.

As I use `Duplicate` also for zones, text and drawings, this will have to be check a bit more thoroughly. And I'll have to think about how to implement the fix so the plugin will be compatible with pre and post 7.0.2 python bindings

BrMisha, 2023-04-22:
Which is the newest version of KiCad where does the ReplicateLayout work?

MitjaNemec, 2023-04-22:
The 2.x.y versions of the plugin should work wit 7.0.1

Tdogb, 2023-04-30:
> new_track = track.Duplicate().Cast()

7.0.0

Tdogb, 2023-04-30:
> Which is the newest version of KiCad where does the ReplicateLayout work?

@BrMisha  7.0.0

MitjaNemec, 2023-05-17:
First of all, thanks for reporting this.

As it turns out this is due to a bug in KiCad. It got introduce just before the 7.0.2 release with [8e579ec8](https://gitlab.com/kicad/code/kicad/-/commit/8e579ec8e0e8d485645d18cb5d897fc2174307c5) and got fixed just after the 7.0.2 was tagged in [7843e6a7](https://gitlab.com/kicad/code/kicad/-/commit/7843e6a70904e9960114a0dc6cfff816b27c116f).

Anyhow, the fix for this is in [2.0.12](https://github.com/MitjaNemec/ReplicateLayout/releases/tag/2.0.12) release, which is available for installation from file and should be available in PCM in a couple of days.

closed 2023-05-23 by MitjaNemec