MitjaNemec/ReplicateLayout #52: Does not replicate traces in KiCad 7.02

TchDuke, 2023-04-24:
[image]
[image]
[replicate_layout.log](https://github.com/MitjaNemec/ReplicateLayout/files/11307525/replicate_layout.log)

MitjaNemec, 2023-04-24:
This is the same as #49. So if you are in a hurry you can patch it locally. Otherwise you'll have to wait a bit

elektronchika, 2023-04-28:
I have the very same log as posted. Placement is replicated, routing not. 

BTW - great plugin, I use it a lot :)

MitjaNemec, 2023-05-17:
First of all, thanks for reporting this.

As it turns out this is due to a bug in KiCad. It got introduce just before the 7.0.2 release with [8e579ec8](https://gitlab.com/kicad/code/kicad/-/commit/8e579ec8e0e8d485645d18cb5d897fc2174307c5) and got fixed just after the 7.0.2 was tagged in [7843e6a7](https://gitlab.com/kicad/code/kicad/-/commit/7843e6a70904e9960114a0dc6cfff816b27c116f).

Anyhow, the fix for this is in [2.0.12](https://github.com/MitjaNemec/ReplicateLayout/releases/tag/2.0.12) release, which is available for installation from file and should be available in PCM in a couple of days.

closed 2023-05-23 by MitjaNemec