MitjaNemec/ReplicateLayout #55: Failure on start V7/2.0.12

myamigo, 2023-05-18:
I'm starting to look closer at V7 to see how it handles my V6 design. I opened and saved it in V7 so the design files are up to date.

When I run the replicator I get the following...
[image]

**replicate_layout_error.log** contains only:

> SyntaxError('invalid syntax', ('C:\\Users\\me\\OneDrive\\Documents\\KiCad\\7.0\\3rdparty\\plugins\\com_github_MitjaNemec_ReplicateLayout\\replicate_layout.py', 1069, 43, '                new_text = text.Duplicate(.Cast()\n'))

What do you suggest?

myamigo, 2023-05-18:
Looks like line 1069 should be 

`
new_text = text.Duplicate(text.Cast())
`

currently it is...

`
new_text = text.Duplicate(.Cast()
`

MitjaNemec, 2023-05-18:
Yeah I was working to fast yesterday and I got a syntax error in.

The line should be
```
new_text = text.Duplicate().Cast()
```
The fix is in [2.0.13](https://github.com/MitjaNemec/ReplicateLayout/releases/tag/2.0.13)

closed 2023-05-23 by MitjaNemec