MitjaNemec/ReplicateLayout #58: ERROR: Destination footprint "...." is member of a different group.

ModuloFS, 2023-06-19:
Hello,

i want to use this plugin to replicate a layout of three hirachial sheets (Optoboard_1 to _3). If start the replication process i an error which says: "Destination footprint Footprint (ref='J2' ---data of footprint---) is member of a different group. All destination plugin hav either have to be members of destination group or no group at all."
The screenshot of the error message as well as the log of the plugin is in the .zip file below. There is also the KiCAD project included.
I have no groups in my schematic or layout. I do not know how to proceed in this problem. Hope somebody could help in this topic.

Thank you!

KiCAD Projekt, Error Message Screenshot and Log:

[Optocoupler_Connectionboard.zip](https://github.com/MitjaNemec/ReplicateLayout/files/11789252/Optocoupler_Connectionboard.zip)

MitjaNemec, 2023-06-19:
Thanks for reporting. Yeah there is a bug with the recently added group feature. I am working on publishing a fix

MitjaNemec, 2023-06-19:
The fix is in 2.1.3. You can install it from file available at https://github.com/MitjaNemec/ReplicateLayout/releases/tag/2.1.3.

This version should be available in PCM in a couple of days

closed 2023-06-19 by MitjaNemec

ModuloFS, 2023-06-20:
Seems to work now! Thank you very much for your fast support!!! This Plugin is awesome!!

commit 62278e94 (2023-06-19): Fix grouping issue

Fixes #58