MitjaNemec/ReplicateLayout #61: Fials with 7.99

pferdob, 2023-07-14:
Execution fails directly after selecting anchor footprint :-(

# Log messsages
```
07-14 14:17:39 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 359:Plugin executed on: 'linux'
07-14 14:17:39 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 360:Plugin executed with python version: '3.11.4 (main, Jun  7 2023, 00:00:00) [GCC 13.1.1 20230511 (Red Hat 13.1.1-2)]'
07-14 14:17:39 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 361:KiCad build version: 7.99.0-1.20230705gitd563e5b.fc38
07-14 14:17:39 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 362:Plugin version: 2.0.10
07-14 14:17:39 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 363:Frame repr: <wx._core.Frame object at 0x7f644ffa52d0>
07-14 14:17:39 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 387:Preparing replicator with Q53 as a reference
07-14 14:17:39 com_github_MitjaNemec_ReplicateLayout.replicate_layout 113:getting a list of all footprints on board
07-14 14:17:39 com_github_MitjaNemec_ReplicateLayout.action_replicate_layout 402:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "/home/frank/.local/share/kicad/7.99/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/action_replicate_layout.py", line 391, in Run
    replicator = Replicator(board, src_anchor_fp_reference)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/frank/.local/share/kicad/7.99/3rdparty/plugins/com_github_MitjaNemec_ReplicateLayout/replicate_layout.py", line 129, in __init__
    sheet_file = fp.GetProperty('Sheetfile')
                 ^^^^^^^^^^^^^^
AttributeError: 'FOOTPRINT' object has no attribute 'GetProperty'
```

# Version
```
Application: KiCad PCB Editor x86_64 on x86_64

Version: 7.99.0-1.20230705gitd563e5b.fc38, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.13.0
	HarfBuzz 7.1.0
	FontConfig 2.14.2
	libcurl/8.0.1 OpenSSL/3.0.9 zlib/1.2.13 brotli/1.0.9 libidn2/2.3.4 libpsl/0.21.2 (+libidn2/2.3.4) libssh/0.10.5/openssl/zlib nghttp2/1.52.0

Platform: Fedora release 38 (Thirty Eight), 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Jul  5 2023 17:19:05
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.78.0
	OCC: 7.6.3
	Curl: 8.0.1
	ngspice: 40
	Compiler: GCC 13.1.1 with C++ ABI 1018

Build settings:
```

pferdob, 2023-07-16:
After more reading, this seems to be a result of the new version not being pushed to the Plugin manager. 

I'll try to manually install the 2.1.x

pferdob, 2023-07-16:
Checked with 2.1.3, sadly does not work.

I the attached project i work on.

[cvtcs-p replicate layout error.zip](https://github.com/MitjaNemec/ReplicateLayout/files/12063435/cvtcs-p.replicate.layout.error.zip)

MitjaNemec, 2023-07-17:
Version 7.99 is not supported by the plugin. You should even not be able to install the plugin via PCM in 7.99

pferdob, 2023-07-17:
Thanks for replying.

I started my first bigger project some months ago using the nigthly build due to improved properties panels and so on. Some 4 weeks ago "suddenly" (unnoticed by me) a change of the on-disk format was committed so that i cannot revert to the stable brnach without losing hundreds of working hours in this projects PCBs.

Can you give me any hint on where to look so that i can try to make a dirty fix myself (still optimistic;-) )? I wonder how to debug these python scripts in conjunction with kicad.

Thanks in any case and for this massively helpful plugin

pferdob, 2023-07-17:
So, it took me two hours tinkering with it, but I succeeded..

There's only three lines to change and it seemes to work for now :)

# Changes

Attention: Right now (nightly from 2023-07-05) undo does not work properly!

## replicate_layout.py

### Line 133
sheet_file = fp.GetSheetfile()  # was fp.GetProperty('Sheetfile')
sheet_name = fp.GetSheetname()  # was fp.GetProperty('Sheetname')

### Line 645

if type(item) is pcbnew.PCB_TEXT:  # if type(item) is pcbnew.FP_TEXT:

pferdob, 2023-07-17:
PS: Did I say thank you for this plugin? It does a decent job with my strange geometry :+1:

MitjaNemec, 2023-07-17:
thanks for trying to fix this yourself and thanks for reporting back.

closed 2023-07-17 by MitjaNemec

MitjaNemec, 2024-01-09:
There is a 7.99+ prerelease available at:https://github.com/MitjaNemec/ReplicateLayout/releases/tag/3.0.1

Currently this is only for installation from file. It'll take me some time to setup the pipline to include it in the PCM