MitjaNemec/ReplicateLayout #66: Brackets in hierarchical sheet title seems to cause them to evade detection

Artem-Sotnikov, 2023-12-01:
I was trying to use the plugin to replicate the layout of hierarchical sheets which had a very convoluted name.

[image]

To achieve this, I installed via the KiCAD manager and selected the tool but none of the destination sheets had shown up.

[image]

I poked at the problem for a bit and renamed the sheets to something simpler on a hunch. When changing them to the following:

[image]

The result is the desired one:

[image]

And the layout gets replicated successfully.

I was able to replicate the issue after re-working through it during the writing of this post for screenshots. It would seem that brackets cause the destination sheets to slip past the plugin. Changing it to a name with brackets retained did not help. Maybe the issue lies elsewhere, but either way this appears to be some sort of undesirable behavior.

Apologies in advance if this turns out to have just been some sort of mistake on my part.

System details:

Windows 11 Home OS build 22621.2715

[image]

[image]

MitjaNemec, 2023-12-01:
Thanks for reporting this. I really appreciate it as it helps making the plugin more robust.

If you have both project still available, can you run the plugin again and store the `replicate_layout.log` file after each run and attach it here. The file should be in the project folder

Additionally if you can prepare a small test project which exhibits the same issue it would be great. Or if you can share this project. And if you'll be preparing test project, feel free to add even more bizarre sheet names (curly braces, square braces, ...) This will spare me the time of preparing a test suite.

It is most likely that the plugin does not handle sheet names properly.

Artem-Sotnikov, 2023-12-01:
Ok, understood! I'll poke around at this when I have time next week and get the logs + example projects.