MitjaNemec/ReplicateLayout #68: Replicating out of scope VIA and tracks

myamigo, 2024-01-29:
I found a new edge case... 

A via with track that originates within the seed replication region that terminates on a component that is NOT in the seed region will be incorrectly replicated when using the "Replicate intersecting tracks/zones/drawings" option.

[VIA_Test.zip](https://github.com/MitjaNemec/ReplicateLayout/files/14087689/VIA_Test.zip) is a very simple example that demonstrates the issue. It consists of three copies of a single sheet which has only 4 test points to define a bounding rectangle on the PCB as well as a single test point at the top that has an attached via that extends into the seed region...

Sub-sheet...
[image]

Top...
[image]

The PCB is as follows...
[image]

After running the replicator with "Replicate intersecting tracks/zones/drawings" enabled I get...
[image]

The via may have tracks extending from it to components that are inside the seed region as well as components that are outside. In this case tracks to the outside components should not be replicated.

Likely related.... UNDO does not remove the incorrectly replicated vias and tracks.

MitjaNemec, 2024-01-29:
Hi, thanks for the report.

I don't see how the plugin works incorrectly
It replicates the contents of the bounding box contents where via and track are connected to the net that is present in the source sheet.
[image]

If you think that the plugin should work differently please describe it more verbosely. Thanks for the persistence

As for the undo, thanks for reporting this. I am porting the plugins to V8 and I've seen that the undo feature does not work on tracks and vias in V8 but it seems that this also changed in V7. If you can vote in the [V8 issue](https://gitlab.com/kicad/code/kicad/-/issues/16590) to increase it's visibility I'd appreciate it. But I'll have to fix the grouping feature which should offer a workaround for this. But I don't know when I'll find the time

MitjaNemec, 2024-01-29:
P.S if you would not check the "intersecting" then only via would be replicated

myamigo, 2024-06-28:
Hi again!

Sorry for not responding sooner. I was working on a design update for some time and finally got through that.

I can use the lock and group attributes to better constrain the design for replication. We can close this issue.

Thank you!

closed 2024-06-28 by myamigo