MitjaNemec/ReplicateLayout #69: KiCAD 8 compatibility

a-allard, 2024-02-09:
Not sure if there is a plan to get this ported to Kicad 8.....

I moved to KiCad 8 which is not supported.  I got this plugin installed and did some debugging to get it running.

The only notable differences is the GetProperty function call is nolonger working so for 

[so line 133](https://github.com/MitjaNemec/ReplicateLayout/blob/62278e94812c5ef6d706d1a8ecc9758a4381b591/replicate_layout.py#L133C17-L134C57) and [line 166](https://github.com/MitjaNemec/ReplicateLayout/blob/62278e94812c5ef6d706d1a8ecc9758a4381b591/replicate_layout.py#L166)

```python
sheet_file = fp.GetProperty('Sheetfile')
sheet_name = fp.GetProperty('Sheetname')
```

becomes

```python
sheet_file = fp.GetSheetfile()
sheet_name = fp.GetSheetname()
```

The only other change I ended up making to get this running was [line 925](https://github.com/MitjaNemec/ReplicateLayout/blob/62278e94812c5ef6d706d1a8ecc9758a4381b591/replicate_layout.py#L925)

```python
src_fp_text_items = self.get_footprint_text_items(src_fp)
dst_fp_text_items = self.get_footprint_text_items(dst_fp)
```

becomes:

```python
src_fp_text_items = src_fp.fp.GetFields() #self.get_footprint_text_items(src_fp)
dst_fp_text_items = dst_fp.fp.GetFields() #self.get_footprint_text_items(dst_fp)
```

And things worked as they had before.  Probably a bit hacked but it works for me running kicad:

Application: KiCad PCB Editor x64 on x64

Version: 8.0.0-rc2-411-g4622738c93, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb  8 2024 06:18:12
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI
Attached is the file I modified to get things running.

[replicate_layout.py.txt](https://github.com/MitjaNemec/ReplicateLayout/files/14222317/replicate_layout.py.txt)

Thank you much for the plugin!  It is really helpful for encouraging good hierarchical design and makes layout much faster.

a-allard, 2024-02-09:
If you want I can do a fork and PR....  Let me know.

MitjaNemec, 2024-02-09:
V8 compatible release is available at: https://github.com/MitjaNemec/ReplicateLayout/releases/tag/3.0.2, the code sits in the 7.99 branch https://github.com/MitjaNemec/ReplicateLayout/tree/7.99

closed 2024-02-09 by MitjaNemec