MitjaNemec/ReplicateLayout #70: Support KICAD 8.0

HaydenHu, 2024-03-05:


MitjaNemec, 2024-03-05:
Already supported in https://github.com/MitjaNemec/ReplicateLayout/releases/tag/3.0.2

HaydenHu, 2024-03-05:
> Already supported in https://github.com/MitjaNemec/ReplicateLayout/releases/tag/3.0.2
[image]
The PCB is a project created by kicad7, which is saved as the kicad8 version, and an error message is reported when it is run

HaydenHu, 2024-03-05:
[image]
I tested your PCB project file and it doesn't work as well
[image]
[image]
[image]

MitjaNemec, 2024-03-05:
This can be something else. Did you just open the layout or the schematics also? Did you try opening the schematics and then updating layout from schematics? In any case, I'd appreciate if you attach the `replicate_layout.log` file. Are you in a position that you could share the project?

HaydenHu, 2024-03-05:
> 这可能是别的东西。您是否也打开了布局或原理图？您是否尝试过打开原理图，然后从原理图更新布局？无论如何，如果您附上文件，我将不胜感激。您是否能够共享该项目？`replicate_layout.log`

[yaokong.zip](https://github.com/MitjaNemec/ReplicateLayout/files/14492649/yaokong.zip)

MitjaNemec, 2024-03-05:
Thanks for sharing. It'll be a day or two before I can get the time to dive into the issue, so please be patient.

HaydenHu, 2024-03-06:
Using kicad8.0, I created the layered schematic diagram and PCB in Chinese language and because of the language respectively, after testing, the results are similar to the test engineering results you provided, all footprints will run around, and the correct results will not be obtained

MitjaNemec, 2024-03-06:
Sorry but we seem to have language issue, so I'd like you to confirm/deny the following:
1. You've created two project as requested with different language settings?
2. Both projects behave the same?
3. With both projects you can see the `sheetfile` and `sheetname` properties within the layout file?

If answer to all of the above questions is "yes" then there is something wrong with you project. possibly there is a bug in KiCad. If the answer to 3 is "no" then there is an issue with your KiCad setup or a bug in KiCad.

Additionally, can I share the project you've attached?

HaydenHu, 2024-03-07:
> 对不起，我们似乎有语言问题，所以我希望您确认/否认以下内容：
> 
> 1. 您是否已根据要求使用不同的语言设置创建了两个项目？
> 2. 两个项目的行为是一样的吗？
> 3. 对于这两个项目，您可以看到布局文件中的 和 属性吗？`sheetfile``sheetname`
1.yes
2.yes
3.The new project in kicad8 has 'sheetfile' and 'sheetname', but although the plug-in is running, it cannot get the correct result

HaydenHu, 2024-03-07:
yaokong.zip
This is a file created by kicad6.0, and an error is reported. The new files created in kicad8 will work with your plug-in

solverstin, 2024-03-25:
Forgive me if my comment is out of topic, but I've tried to install this plugin on KiCAD 8.99 (build locally from gitlab repository)... and it failed to install (from file) with "version incompatible with KiCAD" error. 

Did anybody have that same experience? Is there anything I could do to help bring up this plugin to v8.99 compatibility?

MitjaNemec, 2024-03-25:
Due to following reasons:
- this is likely a KiCad bug
- this is really an unlikely edge case
- lack of info to reproduce this issue
I am closing this issue

closed 2024-03-25 by MitjaNemec