MitjaNemec/ReplicateLayout #72: Texts within footprints are misplaced after replicate layout

showier-drastic, 2024-04-26:
[bug-demo.zip](https://github.com/MitjaNemec/ReplicateLayout/files/15125237/bug-demo.zip)

Before replication:

[image]

After replication:

[image]

Steps to reproduce:

* Open the attached bug demo project.
* Select the LQFP chip U1 (in the already layout area).
* Launch replicate layout tool, use default options.
* Check all the LED modules (D7-D24) and note that all texts are in the wrong place.

showier-drastic, 2024-04-26:
Also, on this project, you can reproduce https://github.com/MitjaNemec/ReplicateLayout/issues/15 when you press Ctrl+Z.

MitjaNemec, 2024-04-26:
Whoa, this really should not happen. It looks like KiCad internals changed a bit with V8. As it will be a while before I can test it much less fix it, can you try running the plugin and disabling replication of text items. If you report back what the results are I'd appreciate it.

showier-drastic, 2024-04-26:
Still happens with text option disabled

[image]

MitjaNemec, 2024-04-26:
Thanks for the prompt response. Can you retry with drawings disabled also?

showier-drastic, 2024-04-26:
Same with drawings & texts disabled.

MitjaNemec, 2024-04-26:
Whoa, this is something seriously wrong. By the look of it it seem that Kicad has reverted to the old style of footprint text positioning. So if I look at the V7 and V8 plugin releases this was the difference I had to indroduce to correct for KiCad internal changes
```
@@ -933,3 +988,3 @@
                         dst_text.SetPosition(dst_txt_pos)
-                        dst_text.SetTextAngleDegrees(-src_txt_orientation)
+                        dst_text.SetTextAngleDegrees(-src_txt_orientation - anchor_delta_angle)
                         dst_text.SetMirrored(not src_text.IsMirrored())
@@ -939,17 +994,5 @@
                         dst_text.SetPosition(dst_txt_pos)
-                        dst_text.SetTextAngleDegrees(src_txt_orientation)
+                        dst_text.SetTextAngleDegrees(src_txt_orientation - anchor_delta_angle)
                         dst_text.SetMirrored(src_text.IsMirrored())
 
```
notice the addition of `anchor_delta_angle`.

I suspect that this has been changed sometime after the KiCad 8 has been released. Which will make it PITA to test as I certainly don't have the same KiCad version as you do.

Anyhow, if you feel courageous, you can find the `replcate_layout.py` file and change the lines 989 and 995 from
```
                        dst_text.SetTextAngleDegrees(-src_txt_orientation - anchor_delta_angle)
```
to 
```
                        dst_text.SetTextAngleDegrees(-src_txt_orientation)

```

If you do please report back the results. I'll look at the KiCad internal changes to figure out when did this change and how to properly solve it. But as I've already said, it'll take quite some time before I do

MitjaNemec, 2024-04-26:
this is jas a not for me:

the likely source of the bug is that `self.get_footprint_text_items()` does not return the same ordered list for source and destination footprints.

MitjaNemec, 2024-05-14:
@showier-drastic this should be solved in [3.0.3](https://github.com/MitjaNemec/ReplicateLayout/releases/tag/3.0.3). The release should be available within PCM in a couple of days

Again, thanks for reporting this. Either the bug did no show as most users' footprints have only a few text items or KiCad's internals changed and text items that I get are not sorted anymore. anyhow, this should not be an issue anymore

showier-drastic, 2024-05-18:
Yes, after updating to 3.0.3, it works great. Thanks for your hard work!

closed 2024-05-18 by showier-drastic

MitjaNemec, 2024-05-18:
Thanks for the feedback and for appreciation.

commit a5735ccf (2024-05-14): sort footprint text items, refactor some parts

fixes #72