MitjaNemec/ReplicateLayout #75: Update replicated group, respect orientation

shufps, 2024-08-02:
What is really missing is to update previously replicated groups and respect the group orientation.

It should update the group without error message and the group and the parts inside of the group should keep it's orientation.

MitjaNemec, 2024-08-02:
Yeah the grouping feature is quite lacking robustness and features. But for the time being, I am lacking time and motivation to do anything about it. Moreover so as python API is going to be removed in the future.

Thanks for understanding

shufps, 2024-09-11:
> Moreover so as python API is going to be removed in the future.

Wait what? :flushed:  Why would they do this? I would never ever again update KiCad then ...