MitjaNemec/ReplicateLayout #76: Typer error 'SwigPyObjec

roneydua, 2024-08-02:
I have this problem
[image]
My current version is:

Application: KiCad x86_64 on x86_64

Version: 8.99.0-unknown-a7d493a6a5~180~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.18

Platform: Ubuntu 22.04.4 LTS, 64 bit, Little endian, wxGTK, X11, ubuntu-xorg, x11
OpenGL: Intel, Mesa Intel(R) Graphics (ADL GT2), 4.6 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2

Build Info:
	Date: Aug  1 2024 04:05:58
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.81.0
	ngspice: 42
	Compiler: GCC 11.4.0 with C++ ABI 1016

Locale: 
	Lang: pt_BR
	Enc: UTF-8
	Num: 1.234,5

Thanks for help

MitjaNemec, 2024-08-02:
Thanks for reporting this. 

I am currently supporting only KiCad V8.0.x. And from the error message and your system description it seems that python API changed in 8.99. 

But from the code inspection it seems that line 416 and 417 are there as a remainder of one of my test. If you can edit the `action_replicate_layout.py` and comment/delete these two lines the plugin should work. When I'll post corrected version I'll let you know

roneydua, 2024-08-02:
I had already edited these lines, but the plugins don't work. I'm sharing the log after the line comment. This plugin is fantastic. I regret having used this nightly version and losing support for this plugin.
[replicate_layout.log](https://github.com/user-attachments/files/16471197/replicate_layout.log)

roneydua, 2024-08-02:
Apparently the addition and subtraction of vectors is different in pcbnew. I was tracking down the code errors and I share here a version that is working.
I solved it by modifying the vectors
a = b- c
making
a = b // to create the vector
a.x = b.x - c.x // manual subtraction of components
a.y = b.y - c.y // manual subtraction of components
in a similar way to addition

roneydua, 2024-08-02:
[modificated_replicate_layout.txt](https://github.com/user-attachments/files/16472173/modificated_replicate_layout.txt)
I'm share the modificated file or replicate_layout.py. The development can compare with the current official.
isnt the best way to solve the problem. I kow