MitjaNemec/ReplicateLayout #80: NameError: name 'src_fp' is not defined

novakma7, 2024-12-10:
When running the ReplicateLayout plugin I am receiving the following error. Log file is attached.

  File "C:\Users\Martin\Documents\KiCad\8.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\action_replicate_layout.py", line 252, in on_ok
    self.replicator.replicate_layout(self.src_anchor_fp, self.src_anchor_fp.sheet_id[0:level + 1],
  File "C:\Users\Martin\Documents\KiCad\8.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 326, in replicate_layout
    self.replicate_footprints(settings)
  File "C:\Users\Martin\Documents\KiCad\8.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 890, in replicate_footprints
    dst_fp = self.match_fp_in_list(src_fp, dst_footprints)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Martin\Documents\KiCad\8.0\3rdparty\plugins\com_github_MitjaNemec_ReplicateLayout\replicate_layout.py", line 853, in match_fp_in_list
    raise LookupError("Can not find destination footprint for source footprint: " + repr(src_fp.ref)
                                                                                         ^^^^^^
NameError: name 'src_fp' is not defined

[replicate_layout.log](https://github.com/user-attachments/files/18080077/replicate_layout.log)

denyofservice, 2025-02-20:
Same error on Linux. First 5-7 times it works fine, then the same error.

[replicate_layout.log](https://github.com/user-attachments/files/18894103/replicate_layout.log)

petersn, 2025-03-06:
Also note that if you replace the offending line 853 of replicate_layout.py with:
```python
                raise LookupError("Can not find destination footprint for source footprint: " + repr(footprint.ref)
```
... then it fixes the error message, and you can properly see which footprint it can't find appropriate matches for.