MitjaNemec/ReplicateLayout #83: kicad 9 compatibility

zodoczi, 2025-03-05:
Hi, I tried to install this on kicad 9 and got this error message :( Can you please check this?

Thank you very much!

[image]

zodoczi, 2025-03-06:
Update: installed the plugin externally, however it crashes when executing: 
[image]

claudiolorini, 2025-03-06:
you could try the things listed [here.](https://github.com/MitjaNemec/ReplicateLayout/pull/82)

zodoczi, 2025-03-07:
Also I wonder is it possible to add / remove layout tracks before cloning? I can manipulate the selection with ctrl+left click, however its not the "white" selection color of the plugin box boundary items, but rather the simple kicad selection color, and thus not inculed in the cloning process. thanks!

MitjaNemec, 2025-04-02:
The plugin has now been release for KiCad V9